aplavins/Arduino-MPPT #1: Inductor

juan3211, 2015-06-12:
Hi,

Thanks a lot !!!! Waht about and schema or similar ? I have check other MPPT in instructables and tim lonan, but yours is for 250W, I have a doubt, what inductor do you use?

Best regards

aplavins, 2015-06-14:
I wound my own 20uH ferrite core inductor with 14awg magnet wire. Schematic
is coming soon!

aplavins, 2015-06-15:
schematic is up
https://github.com/aplavins/Arduino-MPPT/blob/master/MPPT%20charge%20controller%20barebones.pdf

aplavins, 2015-06-15:
Here are some links that will allow you to build your own inductor

http://www.ebay.com/itm/22mm-x-14mm-x-8mm-Round-Green-Toroid-Ferrite-Cores-10-Pcs-/271566883360

http://www.ebay.com/itm/Magnet-Wire-14-AWG-Gauge-Enameled-Copper-4oz-155C-20ft-Magnetic-Coil-Winding-/261085726306?pt=LH_DefaultDomain_0&hash=item3cc9ec0662

You can also salvage these components from used power electronics

use this calculator to find out what value inductor you need:
http://www.daycounter.com/Calculators/Switching-Converter-Calculator.phtml

and this one to find the number of turns for the core and wire you have:
http://coil32.net/online-calculators/ferrite-torroid-calculator.html

Hope this helps!

closed 2015-06-15 by aplavins