aplavins/Arduino-MPPT #3: Low side mosfet burns when the battery is connected

Daniel-Blankenship, 2020-03-19:
Hello,

I attempted to build this project but the low side mosfet gets fried immediately when I connect the battery. I took note of your warning that setting the pulseWidth to a low value will short it out. The main thing I did different with mine was that I powered my Arduino with a usb power bank for testing purposes. I was afraid of connecting vin to the charging battery in case of a voltage spike. I connected the usb power to the arduino first, and then when I connected the charging battery, the low side mosfet popped like a firecracker. The arduino was fine though, and none of the other mosfets seemed damaged according to some testing with a multi-meter.

I have a few assumptions, and please do correct me if I'm wrong, I'm still somewhat inexperienced with these type of electrical projects.

Given that the mosfet burns if the pulse width is set too low, I assume if the IR2104 isn't working then it would also cause this to happen. As far as I understand, it has to be always running or else the low side mosfet makes a closed circuit through the inductor and battery which causes it to fry. I could be wrong though. This makes me think of another potential issue, if the arduino is powered from the 12v battery, would the mosfet cook before the arduino could start up its pwm output to the mosfet driver?

Would adding a diode between the low side mosfets drain and the inductor prevent this burning? Maybe I don't know something that would make adding a diode there a bad idea.

I suspect that my circuit board wiring could have been wrong or maybe a faulty IR2104 that wasn't driving the low side mosfet. I did not have any panels or loads connected when I tested this, I only got to attaching the 12v battery before I had trouble. I replaced the fried mosfet with a new one, thinking that the fried one might have been bad, but the replacement fried too. I'm going to give this another try with a fresh set of components and organize my layout of them better in case I made any mistakes following the schematic.

Thanks for all the work you put into this project, if you have any suggestions please let me know.

aplavins, 2020-03-19:
You need to power the arduino from the battery that you are charging. That's how it gets it's reference voltages, and sets the PWM. It's also possible the IR2104 is faulty.

Daniel-Blankenship, 2020-03-21:
I discovered that I made a dumb mistake haha. The IR2104 chips I have, had a circular indention on the underside in one of the bottom corners that I mistook for being the pin 1 indicator. The actual indicator was printed on the top, on the opposite corner, so I had the chip connected backwards which likely contributed to my mosfet going up in smoke.

Also this time I powered my arduino off the battery as you explained. Even though 12v is recommended, I learned that they could potentially handle up to 20v if the voltage regulator isn't being drawn from too heavily. I was concerned that the lead acid battery when charged could possibly reach 14-15 volts at its highest but it seems that this isn't a problem.

I put together a stripped down version of this circuit on a breadboard for testing ir2104's, it just uses an led for the load instead of running an inductor coil. I made some test code and everything worked perfectly. The led changes brightness accordingly as I modify the pwm values through 75-245, making sure to disable the driver during changes.

I also educated myself a bit more about ir2104 half bridges, it seems its fine not immediately having pwm from the arduino when power is first received, which makes more sense than what I speculated about it starting up the pwm my first post. I later tested this by powering it without the arduino connected and nothing melted.

I appreciate you pointing me in the right direction, thank you for your help!

[image]
[image]

svdrummer, 2020-03-22:
Thank you for following up with this. Good feedback for us all.

closed 2020-04-03 by Daniel-Blankenship